slaclab/rogue #823: Bad access to shared pointer when creating UDP client and server

maTizK, 2021-12-07:
Constructor of `rogue::protocols::udp::Client` and `rogue::protocols::udp::Server` spawns a `runThread`. Inside this thread a `shared_from_this()` is called, that returns a shared pointer of the constructed object. 
In the case where the object is already constructed, e.g. shared pointer is not weak it works fine.
However, if the spawned thread `runThread` tries to generate a shared pointer before the object is constructed it throws `std::bad_weak_ptr`. 

Now, it seems that `runThread` of `Client` and `Server` has a `usleep(1000);` to overcome this problem. However, this is not always sufficient. The `runThread` shall wait until the object is created, i.e. shared pointer was created out of it. 

For example with use of mutex
```cpp
//! Class creation
rpu::ServerPtr rpu::Server::create(uint16_t port, bool jumbo)
{
   rpu::ServerPtr r = std::make_shared<rpu::Server>(port, jumbo);
   r->constructed();
   return (r);
}

//! Set object is constructed
void rpu::Server::constructed()
{
   mutex_.unlock();
}

//! Run thread
void rpu::Server::runThread()
{
   ris::BufferPtr buff;
   ris::FramePtr frame;
   fd_set fds;
   int32_t res;
   struct timeval tout;
   struct sockaddr_in tmpAddr;
   uint32_t tmpLen;
   uint32_t avail;

   udpLog_->logThreadId();
   std::lock_guard<std::recursive_mutex> lk(mutex_); // wait for construction to end

   // Preallocate frame
   frame = ris::Pool::acceptReq(maxPayload(), false);

   while (threadEn_)
   {
```

slacrherbst, 2021-12-07:
Jesus and I could not figure out a solution to this.  We have this problem in numerous places:

https://github.com/slaclab/rogue/commit/2b4358cf5107b7fbcdf272b6c71c8ab6ef80b1a0

This appears to be a clever solution to this.

maTizK, 2021-12-07:
Maybe this example helps to solve the issue since locking mutex introduces a new possible deadlock. In this example it's required to call `destroy()` or in your case `stop()`. With this you avoid any undefined behaviour if destructor gets called in between construction and spawning a thread. 

```cpp
#include <chrono>
#include <future>
#include <iostream>
#include <memory>
#include <string>
#include <thread>
#include <unistd.h>
#include <vector>
#include <date/date.h>

struct Logger
{

  template <typename Arg0>
  void log(Arg0 const &arg)
  {
    std::lock_guard<std::recursive_mutex> lk(m_mutex);
    using namespace date;
    std::cout << arg << " ";
  }

  template <typename Arg0, typename... Args>
  void log(Arg0 const &arg0, Args const &...args)
  {
    std::lock_guard<std::recursive_mutex> lk(m_mutex);
    using namespace date;
    std::cout << arg0;
    if constexpr (sizeof...(Args) > 0U)
    {
      Logger::log(args...);
    }
  }

  std::recursive_mutex m_mutex;
};

static Logger logger;

#define LOG(...)                                                                                               \
  {                                                                                                            \
    using namespace date;                                                                                      \
    logger.log("[", std::chrono::system_clock::now(), "]\t", __func__, ":", __LINE__, ":", __VA_ARGS__, "\n"); \
  }

class EnableSharedFromThisBase
    : public std::enable_shared_from_this<EnableSharedFromThisBase>
{
public:
  virtual ~EnableSharedFromThisBase()
  {
    LOG("Calling ~EnableSharedFromThisBase()");
  };
};

template <typename T>
class EnableSharedFromThis : virtual public EnableSharedFromThisBase
{
public:
  std::shared_ptr<T> shared_from_this()
  {
    return std::dynamic_pointer_cast<T>(
        EnableSharedFromThisBase::shared_from_this());
  }
};

struct B : EnableSharedFromThis<B>
{

  void get() { std::cout << "Running get function\n"; }

  virtual ~B()
  {
    LOG("Calling ~B()");
  };
};

struct A : B
{

  A()
  {
    m_mutex.lock();
    thread_ = new std::thread(&A::runThread, this);
    std::cout << "End of A::A()\n";
  }

  static auto create()
  {
    LOG("Creating A");
    auto ptr = std::make_shared<A>();
    LOG("Created A");
    ptr->constructed();
    return ptr;
  }

  void constructed()
  {
    m_mutex.unlock();

    LOG("Mutex unlocked");
  }

  void runThread()
  {
    LOG("Running thread");
    std::unique_lock<std::recursive_mutex> lk(m_mutex);
    LOG("Got Mutex");

    auto ptr = shared_from_this();
    ptr->get();
    std::cout << "End of A::runThread()\n";
    lk.unlock();
  }

  void destroy()
  {
    thread_->join();
    m_destroyed = true;
  }

  ~A()
  {
    LOG("Calling destructor");
    m_destroyed.load() ? nullptr : throw std::runtime_error("Call destroy() before destructor");
    delete thread_;
  }

  std::thread *thread_;
  std::recursive_mutex m_mutex;
  std::atomic<bool> m_destroyed{false};

private:
};

int main()
{
  {
    auto a = A::create(); // creatas A
    // LOG("shared count", a.use_count());
    a->get();

    a->destroy();
  }
{
    auto a = A::create(); // creatas A
    // LOG("shared count", a.use_count());
    a->get();

}
}
```

ruck314, 2022-01-12:
Refer to https://github.com/slaclab/rogue/pull/826 for proposed fix

mskoufis, 2022-02-07:
New proposed fix for this is #836

closed 2022-02-14 by mskoufis