slaclab/rogue #863: UDP Server and Client internal thread may cause std::bad_weak_ptr exception to be thrown

PrimozLavric, 2022-04-07:
shared_from_this() may be called before the object and shared_ptr are fully initialized. This causes std::bad_weak_ptr exception and crashes the application.

Trace Server:
[rpu::Server::runThread()](https://github.com/slaclab/rogue/blob/main/src/rogue/protocols/udp/Server.cpp#L87)
-- [ris::Pool::acceptReq(maxPayload(),false)](https://github.com/slaclab/rogue/blob/main/src/rogue/protocols/udp/Server.cpp#L190)
---- [allocBuffer(size,&frSize)](https://github.com/slaclab/rogue/blob/main/src/rogue/interfaces/stream/Pool.cpp#L74)
------ [shared_from_this()](https://github.com/slaclab/rogue/blob/main/src/rogue/interfaces/stream/Pool.cpp#L166)

Trace Client:
[rpu::Client::runThread()](https://github.com/slaclab/rogue/blob/main/src/rogue/protocols/udp/Client.cpp#L90)
-- [ris::Pool::acceptReq(maxPayload(),false)](https://github.com/slaclab/rogue/blob/main/src/rogue/protocols/udp/Client.cpp#L186)
---- [allocBuffer(size,&frSize)](https://github.com/slaclab/rogue/blob/main/src/rogue/interfaces/stream/Pool.cpp#L74)
------ [shared_from_this()](https://github.com/slaclab/rogue/blob/main/src/rogue/interfaces/stream/Pool.cpp#L166)

Proposed fix:
```
auto weakPtr = weak_from_this();
auto sharedPtr = weakPtr.lock();

while (!sharedPtr) {
    std::this_thread::yield();
    sharedPtr = weakPtr.lock();
}
```

bengineerd, 2022-04-19:
This might have been fixed by https://github.com/slaclab/rogue/pull/836

PrimozLavric, 2022-04-19:
Yes, that indeed fixes the issue.

closed 2022-04-19 by ruck314